TempleRAIL/drl_vo_nav #7: the position of the pedestrian drift

SiyiLoo, 2023-09-11:
when I roslaunch drl_vo_nav_train.launch and the drl_vo_train.py node is started, [the position of the pedestrian will drift may be after resetting the initial position of the robot. [https://file.notion.so/f/f/a9e31244-4ace-460b-9930-c012bef7c1e7/93af9cb5-ae9c-4e96-a35f-6111cadff4d1/1.png?id=7a3d0152-3b3a-4d02-a079-09dc3785b8bb&table=block&spaceId=a9e31244-4ace-460b-9930-c012bef7c1e7&expirationTimestamp=1694484000000&signature=k2Ui6E_vSgNkNlx7-Nb9kvI_qXkfmhP9J32x-9bc2OE&downloadName=1.png](url)

zzuxzt, 2023-09-11:
Although I cannot see the picture you shared, I don't think the slight drift in the Rviz during the reset phase is a big problem. The Rviz visualization of pedestrians is based on the TF of the robot. The slight drift is due to the inaccuracy of the given initial position for the robot, and the AMCL package will automatically correct the robot's position later.

SiyiLoo, 2023-09-19:
There is static transformer between /odom and /gazebo coordinate system in robot.launch(<node pkg="tf" type="static_transform_publisher" name="gazebo_map_broadcaster"
      args="-1 -1 0 0 0 0  odom gazebo 100"/>), but this static transformer only useful when robot's initial position is (1,0,0.13). So when I change robot's initial position,  the position of the pedestrian will drift. Could you please tell me how the translation and rotation relation in static transformer is determined? Thank you very much, your reply will be of great help to me.

zzuxzt, 2023-09-21:
Thanks for giving more details. I also modified the Rviz file to let it show correct pedestrian visualization, please update the [robot_gazebo](https://github.com/TempleRAIL/robot_gazebo) package. 
The static transform node is a useless node. But as I said before, pedestrian position drift in Rviz is just a visualization drift, caused by errors in robot localization. The simulator follows the real-world imperfect settings, where the localizatin service is imperfect and provided by the AMCL package. If you want to change the initial position of the robot in the Gazebo world and see the correct Rviz visualization, you need to give a correct rough initial position to the AMCL on Rivz (assuming you are operating the robot in the real world and need to localize it, the first step is to give a rough initial point). Hope it solves your doubts.

closed 2023-10-02 by zzuxzt