TempleRAIL/drl_vo_nav #8: Can not get robot pose

YumengXiu, 2023-09-21:
Hello there,
When I do the training running  `sh run_drl_vo_policy_training_desktop.sh`, all nodes start, but the terminal shows "cannot get the robot pose" and robot does not move in gazebo.
How can I fix it?

zzuxzt, 2023-09-22:
Well, I think you just have to wait longer for the training code to run. At the start, it will show the error "cannot get the robot pose". If it does not work, just restart it again. It should work then.

YumengXiu, 2023-09-22:
Hi,
Thanks for the answer, looks like the desktop training still does not run even when I restart it, the robot did not move in rviz/gazebo. Here are the warning messages in the terminal:
[image]

Am I missing anything?

zzuxzt, 2023-09-23:
Hi, thanks for finding this issue. I retested it on a new computer and found that the problem was caused by the gym environment not being registered. I updated the bash command files. Please update your drl_vo package code and run it again. It should work this time.

YumengXiu, 2023-09-23:
Hi, many thanks for the swift reply.
Actually I had registered the turtlebot gym environment before running the old bash file, the problem may not be caused by it, but I just tested the new bash file on a new computer and the robot is still not moving, I found another error caused by the "/map" topic, I think this causes the warning message "cannot get obot pose". In rivz it also shows "no map received".
[image]

zzuxzt, 2023-09-23:
Hmm, I did not meet this problem. How did you run the code? Using your PC or the singularity container? By the way, can you run the navigation demo bash file?

If you set up the environment on your PC, can you install the following packages and test it again?
sudo apt-get install ros-noetic-move-base*
sudo apt-get install ros-noetic-map-server*
sudo apt-get install ros-noetic-amcl*
sudo apt-get install ros-noetic-navigation*

YumengXiu, 2023-09-23:
Thanks for the tip! I set up the environment on my PC by noetic, the four packages do help solve the robot_pose issue, while the visualization works fine when I set a random navigation goal on rviz, but the robot still doesn't move even when running navigation demo bash file. Do I need to publish a goal using command line or am I missing other packages?
[image]

zzuxzt, 2023-09-24:
You can directly use the Rviz to publish a goal point. It needs some time to load the DRL-VO model but it should work when the model loading is finished. 
Did you install the Turtlebot 2 packages? By the way, you can try to install the following packages:
```
sudo apt-get install ros-noetic-joy ros-noetic-teleop-twist-joy ros-noetic-teleop-twist-keyboard ros-noetic-laser-proc ros-noetic-rgbd-launch ros-noetic-rosserial-arduino ros-noetic-rosserial-python ros-noetic-rosserial-client ros-noetic-rosserial-msgs ros-noetic-amcl ros-noetic-map-server ros-noetic-move-base ros-noetic-urdf ros-noetic-xacro ros-noetic-compressed-image-transport ros-noetic-rqt-image-view ros-noetic-gmapping ros-noetic-navigation ros-noetic-interactive-markers ros-noetic-pcl-ros
```

YumengXiu, 2023-09-24:
Yes, I've installed the turtlebot2 by "sh turtlebot2_noetic_install.sh", but looks like these packages do not solve the issue, how long will it take to load the DRL-VO model?
By the way, when looking into the "turtlebot2_noetic_install.sh" file, 
`sudo apt-get install ros-noetic-joystick-drivers ` is not available for noetic distro, I'm not sure if this cause the freezing issue.

zzuxzt, 2023-09-24:
The model loading time depends on your computer, it can be several seconds or around 1 minute. When it was loaded, you can see the "Finish loading model." log on your terminal window. 

However, I guess the problem is due to the incomplete installation of turtlebot2 packages. Can you try to manually install all the required packages line by line from the [turtlebot2_noetic_install.sh](https://github.com/zzuxzt/turtlebot2_noetic_packages/blob/master/turtlebot2_noetic_install.sh)? 
Before starting your installation, run the following command first to update your source list:
```
sudo apt-get update 
```
That joystick-drivers package should work but ignore it for now. After installation, you should see the following packages in your workplace:
[image]

After that, you can test if you can control your robot via keyboard using the following command?
```
roslaunch turtlebot_gazebo turtlebot_world.launch
roslaunch turtlebot_teleop keyboard_teleop.launch
```

YumengXiu, 2023-09-25:
I found that the pretrained model was not able to load on my PC, now I resolve the issue. Again, thanks for the great work and help!

closed 2023-09-25 by YumengXiu