TempleRAIL/drl_vo_nav #9: Could not get robot pose

caomuqing, 2023-10-18:
Hi,

Thanks for this package. I am facing a similar issue as this [issue](https://github.com/TempleRAIL/drl_vo_nav/issues/8), but I have read the conversation but still could not resolve the issue.
It appears that all nodes are running successfully, and loading model is finished, there is no error message related to node dying. But there is some warning as shown below:

[image]

The rviz shows no output:
[image]

I can see the gazebo spawns correctly, with the robot and moving people. What could be the issue?

zzuxzt, 2023-10-18:
Have you tried restarting it? It looks like some parts related to the turtlebot2 TF model didn't run successfully. Or can you provide more terminal log information?

caomuqing, 2023-10-19:
Thank you for your reply. I have tried restarting it many times. Here is the log output from running `sh run_drl_vo_navigation_demo.sh` to kill:

```
... logging to /home/iot02/.ros/log/9cfbc5a8-6e36-11ee-abbf-799c719367ea/roslaunch-iotdelta-233059.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://iotdelta:45169/

SUMMARY
========

PARAMETERS
 * /amcl/base_frame_id: base_footprint
 * /amcl/global_frame_id: map
 * /amcl/gui_publish_rate: 10.0
 * /amcl/initial_pose_a: 0.13
 * /amcl/initial_pose_x: 1.0
 * /amcl/initial_pose_y: 0.0
 * /amcl/kld_err: 0.05
 * /amcl/kld_z: 0.99
 * /amcl/laser_lambda_short: 0.1
 * /amcl/laser_likelihood_max_dist: 2.0
 * /amcl/laser_max_beams: 60
 * /amcl/laser_max_range: 12.0
 * /amcl/laser_model_type: likelihood_field
 * /amcl/laser_sigma_hit: 0.2
 * /amcl/laser_z_hit: 0.5
 * /amcl/laser_z_max: 0.05
 * /amcl/laser_z_rand: 0.5
 * /amcl/laser_z_short: 0.05
 * /amcl/max_particles: 2000
 * /amcl/min_particles: 500
 * /amcl/odom_alpha1: 0.2
 * /amcl/odom_alpha2: 0.2
 * /amcl/odom_alpha3: 0.2
 * /amcl/odom_alpha4: 0.2
 * /amcl/odom_alpha5: 0.1
 * /amcl/odom_frame_id: odom
 * /amcl/odom_model_type: diff
 * /amcl/recovery_alpha_fast: 0.0
 * /amcl/recovery_alpha_slow: 0.0
 * /amcl/resample_interval: 1
 * /amcl/transform_tolerance: 1.0
 * /amcl/update_min_a: 0.2
 * /amcl/update_min_d: 0.25
 * /amcl/use_map_topic: False
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /cmd_vel_mux/yaml_cfg_file: /home/iot02/catki...
 * /drl_vo_cmd/model_file: /home/iot02/catki...
 * /gazebo/enable_ros_network: True
 * /move_base/DWAPlannerROS/acc_lim_theta: 2.0
 * /move_base/DWAPlannerROS/acc_lim_x: 1.0
 * /move_base/DWAPlannerROS/acc_lim_y: 1.0
 * /move_base/DWAPlannerROS/forward_point_distance: 0.5
 * /move_base/DWAPlannerROS/global_frame_id: odom
 * /move_base/DWAPlannerROS/goal_distance_bias: 24.0
 * /move_base/DWAPlannerROS/max_scaling_factor: 0.2
 * /move_base/DWAPlannerROS/max_vel_rot: 3.0
 * /move_base/DWAPlannerROS/max_vel_trans: 0.5
 * /move_base/DWAPlannerROS/max_vel_x: 0.5
 * /move_base/DWAPlannerROS/max_vel_y: 0.0
 * /move_base/DWAPlannerROS/min_vel_rot: 0.4
 * /move_base/DWAPlannerROS/min_vel_trans: 0.1
 * /move_base/DWAPlannerROS/min_vel_x: 0.0
 * /move_base/DWAPlannerROS/min_vel_y: 0.0
 * /move_base/DWAPlannerROS/occdist_scale: 0.5
 * /move_base/DWAPlannerROS/oscillation_reset_dist: 0.05
 * /move_base/DWAPlannerROS/path_distance_bias: 64.0
[... 595 lines cut ...]
/bumper2pointcloud
/cmd_vel_mux
/cnn_data_pub
/drl_vo_cmd
/gazebo
/gazebo_gui
/gazebo_map_broadcaster
/goal_visualize
/kobuki_safety_controller
/map_server
/mix_cmd_vel
/mobile_base_nodelet_manager
/move_base
/navigation_velocity_smoother
/pedsim_simulator
/pedsim_visualizer
/pure_pursuit
/robot_pose_pub
/robot_state_publisher
/rosout
/rviz
/teleop_velocity_smoother
/track_ped_pub

```

caomuqing, 2023-10-19:
Also, here is the tf tree obtained from `rosrun rqt_tf_tree rqt_tf_tree`
[image]

zzuxzt, 2023-10-20:
Hi, do you install all packages in your computer instead of using singularity container? 
Seems that the main error is "Couldn't find the name of one of the wheel drop sensors in the model description", which is from the Turtlebot 2 Kobuki base simulation model. The model definition in my packages should be right. Here are my suspections for this error:
1. I suspect this error is caused by an incompatible Gazebo. What is your verison of Gazebo? Make sure that the version of Gazebo are compatible with ROS Noetic, which should be Gazebo V11.x (mine is V11.9.0).
2. The Kobuki ros package was not installed sucessfully. Please refer to [turtlebot2_noetic_packages](https://github.com/zzuxzt/turtlebot2_noetic_packages/blob/master/turtlebot2_noetic_install.sh) to install it.
3. If they do not work, can you help to check the default Turtlebot2 navigation demo with the following commands:
```
roslaunch turtlebot_gazebo turtlebot_world.launch
roslaunch turtlebot_gazebo amcl_demo.launch
roslaunch turtlebot_rviz_launchers view_navigation.launch
```
Please feel free to let me know more details.

caomuqing, 2023-10-20:
Thanks for your hints. Indeed, the problem is caused by the error message "Couldn't find the name of one of the wheel drop sensors in the model description". What I do to resolve this is adding the following sensor setup in the in your file [kobuki_gazebo.urdf.xacro](https://github.com/TempleRAIL/robot_gazebo/blob/master/urdf/kobuki_gazebo.urdf.xacro).
I believe this change is needed for all users who want to run it by self-installation of all packages, so i create a pull request.
Now, the program can run successfully.

zzuxzt, 2023-10-22:
Glad to hear you solved this problem. 
Thanks for your contribution and I will approve it. I just noticed that the kobuki URDF definition has been updated recently, and newer versions of ROS Noetic and its compatible Gazebo require updated URDF definitions.

closed 2023-10-22 by zzuxzt

tkddnjs98, 2023-10-22:
Can you tell me where should I reference the updated URDF definition?

zzuxzt, 2023-10-22:
You can found it at [kobuki_gazebo.urdf.xacro](https://github.com/TempleRAIL/robot_gazebo/blob/2f37936f07974d442ec884c6789530c4a9ea01f1/urdf/kobuki_gazebo.urdf.xacro#L12). Added wheel_drop_left/right contact sensors to the Kobuki. The original source is [here](https://github.com/yujinrobot/kobuki/pull/425/files).

tkddnjs98, 2023-10-22:
Thanks I really appreciate that.