TempleRAIL/drl_vo_nav #10: How to use VO

GXJll, 2023-10-21:
Hello, after reading your paper, I have developed a great interest in VO. Could you please ask which part of the code reflects speed barriers that I have not found, and how did you use this VO in the PPO algorithm? I hope you can help me answer this question. Thank you very much for your reply

zzuxzt, 2023-10-22:
Thank you for your interest in our work. As I worte in the paper, As I wrote in my paper, I designed a heading direction reward function using VO to encourage the robot to actively tune its heading direction to avoid moving pedestrians and move toward the goal point. The detailed implementation are in [theta_reward](https://github.com/TempleRAIL/drl_vo_nav/blob/7a3d8f9be2963ebacdaad4b676aa5de3a8c5d9ce/drl_vo/src/turtlebot_gym/turtlebot_gym/envs/drl_nav_env.py#L1010).

GXJll, 2023-10-22:
thank you for your reply. One more question, is VO input into the neural network as a part of the state?

zzuxzt, 2023-10-22:
No, the input data are the current pedestrian kinemitic maps, lidar historical map, and the sub-goal point. More details can be found in our paper.

closed 2023-10-30 by zzuxzt