TempleRAIL/drl_vo_nav #11: Is camera image used in simulation?

caomuqing, 2023-10-22:
Hi, this is not really an issue but just a question. From the simulation, I don't see any zed camera image or depth data being used, it seems that the pedestrian's information from the pedsim is fed into the network directly. So, the simulation does not include the image encoding part described in the paper?

zzuxzt, 2023-10-22:
Sorry for the inconvenience, the current open-source code does not include the Yolo detector and MHT tracker, and the pedestrian information comes directly from the pedsim_simulator, which makes the simulator run faster. 
We did not open source the Yolo detector and MHT tracker code because the MHT tracker required some commercial license packages, which is a bit complicated. But when we clear these commercial restrictions and update the code, we will consider open sourcing the same preprocessing pipeline as in our paper.

caomuqing, 2023-10-23:
I see, thank you for your information!

closed 2023-10-23 by caomuqing