TempleRAIL/drl_vo_nav #14: The meaning of values contained in /map topic

alantes, 2024-01-09:
Dear researcher,

Thank you for your fantastic work. I am working on running your project. However, the messages received by /map seems not meaningful. In a map that is almost empty, the map.data I received is almost full of -1. If I understand it correctly, value 0 stands for open space without obstacles. So what is the meaning of -1 and 100:

[image]

WIT23, 2024-01-09:
The map is loaded from a pgm format file. 100 means obstacle and -1 means no information.

alantes, 2024-01-09:
Thanks a lot for your clarification!

closed 2024-01-09 by alantes