TempleRAIL/drl_vo_nav #15: drl_vo_demo error

CAI23sbP, 2024-01-09:
How are you @zzuxzt 
i was following your docs, but there is some issue 
here is error script
`[ INFO] [1704766854.412964438, 0.007000000]: Physics dynamic reconfigure ready.
[WARN] [1704766854.443912, 0.036000]: Could not get robot pose
[drl_vo_cmd-24] process has died [pid 10043, exit code 1, cmd /home/cai_gh/catkin_ws/src/drl_vo_nav/drl_vo/src/drl_vo_inference.py __name:=drl_vo_cmd __log:=/home/cai_gh/.ros/log/b56d4608-ae95-11ee-8023-cb9616f8f13b/drl_vo_cmd-24.log].
log file: /home/cai_gh/.ros/log/b56d4608-ae95-11ee-8023-cb9616f8f13b/drl_vo_cmd-24*.log
`

zzuxzt, 2024-01-09:
Hello, can you try re-running again? Sometimes, such errors occur due to Gazebo not fully running. If re-running does not solve your issue, please give the full log information.

CAI23sbP, 2024-01-09:
@zzuxzt thank you for your quick reply
I have waited for starting code. It was work. 
By the way, when i pointed goal position in rviz, a robot didn't move to goal position.
here is screen shot and full log
 
[image]
`... logging to /home/cai_gh/.ros/log/8b56cfcc-aea6-11ee-8023-cb9616f8f13b/roslaunch-cai-gh-46763.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cai-gh:42763/

SUMMARY
========

PARAMETERS
 * /amcl/base_frame_id: base_footprint
 * /amcl/global_frame_id: map
 * /amcl/gui_publish_rate: 10.0
 * /amcl/initial_pose_a: 0.13
 * /amcl/initial_pose_x: 1.0
 * /amcl/initial_pose_y: 0.0
 * /amcl/kld_err: 0.05
 * /amcl/kld_z: 0.99
 * /amcl/laser_lambda_short: 0.1
 * /amcl/laser_likelihood_max_dist: 2.0
 * /amcl/laser_max_beams: 60
 * /amcl/laser_max_range: 12.0
 * /amcl/laser_model_type: likelihood_field
 * /amcl/laser_sigma_hit: 0.2
 * /amcl/laser_z_hit: 0.5
 * /amcl/laser_z_max: 0.05
 * /amcl/laser_z_rand: 0.5
 * /amcl/laser_z_short: 0.05
 * /amcl/max_particles: 2000
 * /amcl/min_particles: 500
 * /amcl/odom_alpha1: 0.2
 * /amcl/odom_alpha2: 0.2
 * /amcl/odom_alpha3: 0.2
 * /amcl/odom_alpha4: 0.2
 * /amcl/odom_alpha5: 0.1
 * /amcl/odom_frame_id: odom
 * /amcl/odom_model_type: diff
 * /amcl/recovery_alpha_fast: 0.0
 * /amcl/recovery_alpha_slow: 0.0
 * /amcl/resample_interval: 1
 * /amcl/transform_tolerance: 1.0
 * /amcl/update_min_a: 0.2
 * /amcl/update_min_d: 0.25
 * /amcl/use_map_topic: False
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /cmd_vel_mux/yaml_cfg_file: /home/cai_gh/catk...
 * /drl_vo_cmd/model_file: /home/cai_gh/catk...
 * /gazebo/enable_ros_network: True
 * /move_base/DWAPlannerROS/acc_lim_theta: 2.0
 * /move_base/DWAPlannerROS/acc_lim_x: 1.0
 * /move_base/DWAPlannerROS/acc_lim_y: 1.0
 * /move_base/DWAPlannerROS/forward_point_distance: 0.5
 * /move_base/DWAPlannerROS/global_frame_id: odom
 * /move_base/DWAPlannerROS/goal_distance_bias: 24.0
 * /move_base/DWAPlannerROS/max_scaling_factor: 0.2
 * /move_base/DWAPlannerROS/max_vel_rot: 3.0
 * /move_base/DWAPlannerROS/max_vel_trans: 0.5
 * /move_base/DWAPlannerROS/max_vel_x: 0.5
 * /move_base/DWAPlannerROS/max_vel_y: 0.0
 * /move_base/DWAPlannerROS/min_vel_rot: 0.4
 * /move_base/DWAPlannerROS/min_vel_trans: 0.1
 * /move_base/DWAPlannerROS/min_vel_x: 0.0
 * /move_base/DWAPlannerROS/min_vel_y: 0.0
[... 399 lines cut ...]
[ WARN] [1704774087.879175332, 0.288000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint at time 0.288000 according to authority unknown_publisher
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint at time 0.288000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint at time 0.292000 according to authority /gazebo
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint at time 0.292000 according to authority /gazebo
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint at time 0.292000 according to authority /gazebo
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint at time 0.292000 according to authority /gazebo
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint at time 0.292000 according to authority /gazebo
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint at time 0.292000 according to authority /gazebo
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint at time 0.292000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint at time 0.292000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint at time 0.292000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp
[ WARN] [1704774087.892836868, 0.295000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint at time 0.292000 according to authority unknown_publisher
[ WARN] [1704774087.892868132, 0.295000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint at time 0.292000 according to authority unknown_publisher
[ WARN] [1704774087.892885534, 0.295000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint at time 0.292000 according to authority unknown_publisher
`

CAI23sbP, 2024-01-09:
Sorry for @zzuxzt .
I am really sorry, above issue is my mistake...
Because of Stable-baselines3's compatibility issue which was a protobuf package.
when i re-installed protobuf package's downgrade version it was solved

closed 2024-01-09 by CAI23sbP

zzuxzt, 2024-01-09:
Based on your log error, 
```
TypeError: Descriptors cannot be created directly.
If this call came from a _pb2.py file, your generated code is out of date and must be regenerated with protoc >= 3.19.0.
If you cannot immediately regenerate your protos, some other possible workarounds are:

Downgrade the protobuf package to 3.20.x or lower.
Set PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python (but this will use pure-Python parsing and will be much slower).
```
you can try to install the protobuf package: 
```
pip install protobuf==3.20.*
```