TempleRAIL/drl_vo_nav #16: Question: How to use another map?

CAI23sbP, 2024-01-09:
@zzuxzt , Hi!
I have question that How to use another map?
In your paper there were existing 5 maps.
 
[image]
I wanted to use it, so i searched your code which accepted another map.
`  <arg name="scene_file" default="$(find pedsim_simulator)/scenarios/eng_hall.xml"/>
  <arg name="world_name" default="$(find pedsim_gazebo_plugin)/worlds/eng_hall.world"/>
  <arg name="gui" default="true" doc="Bring up the Gazebo graphical interface"/>
  <arg name="pose_initial_x" default="1.0"/> 
  <arg name="pose_initial_y" default="1.0"/>
  <arg name="map_file" default="$(find robot_gazebo)/maps/gazebo_eng_lobby/gazebo_eng_lobby.yaml"/>
  <arg name="initial_pose_x" default="1.0"/>
  <arg name="initial_pose_y" default="0.0"/>
  <arg name="initial_pose_a" default="0.13"/>
`
When i changed map_file, world name, scene file.but map is not changing in gazebo, and robot init position. could you tell me how to change map?

zzuxzt, 2024-01-10:
Yeah, you need to change the scene_file, world_name, initial_pose, map_file in "[drl_vo_nav.launch](https://github.com/TempleRAIL/drl_vo_nav/blob/7a3d8f9be2963ebacdaad4b676aa5de3a8c5d9ce/drl_vo/launch/drl_vo_nav.launch#L2)" file to do that. For example, the below configurations can change the map to the Autolab environment:
```
  <arg name="scene_file" default="$(find pedsim_simulator)/scenarios/autolab/autolab_25.xml"/>
  <arg name="world_name" default="$(find pedsim_gazebo_plugin)/worlds/autolab.world"/>
  <arg name="gui" default="true" doc="Bring up the Gazebo graphical interface"/>
  <arg name="pose_initial_x" default="1.0"/> 
  <arg name="pose_initial_y" default="1.0"/>
  <arg name="map_file" default="$(find robot_gazebo)/maps/gazebo_autolab/gazebo_autolab.yaml"/>
  <arg name="initial_pose_x" default="0.0"/>
  <arg name="initial_pose_y" default="0.0"/>
  <arg name="initial_pose_a" default="0.0"/>
```

tkddnjs98, 2024-01-13:
Another Question I tried using other robot and I found something interesting.
What are the roles of "nodelet" and "yocs velocity smoother?"
I think the system takes both of DWAlocalplanner's output and drl_vo's output but finally sends drl_vo's output to the robot.

zzuxzt, 2024-01-15:
"nodelet" is used to implement multiplexing of multiple control topics (e.g. teleop cmd and move_base cmd). The "yocs velocity smoother" is used to smooth the raw control commands. Please take a look at the ROS introduction websites to know more details.
 
The DWA output was remapped to dwa_cmd (a lazy trick for not writing a new specific local planner), which was not used and was abandoned. Note that the system is only controlled by the drl_vo_cmd. If you wanna use the other robot, please remap the drl_vo_cmd to the control topic of that robot. One example can be found in my [BARN-challenge repository.](https://github.com/TempleRAIL/nav-competition-icra2022-drl-vo.git)

tkddnjs98, 2024-01-15:
Thanks I really appreciate your comment. I'll take a look about it.

closed 2024-01-17 by zzuxzt