TempleRAIL/drl_vo_nav #21: Implementing an other robot differential robot instead of Turtlebot2

FurkanEdizkan, 2024-02-16:
Hello again,

Following up on issue #19, I've implemented a new differential robot using Gazebo plugins for drive, LiDAR, and other functionalities. While training runs successfully, the robot itself remains stationary.

I've confirmed that the scan code feeds correctly into ROS. However, the differential drive plugin isn't receiving any cmd_vel messages. I've examined the code and seen that cmd_vel is often remapped to different topics depending on specific requirements, but I haven't been able to pinpoint the exact topic I need for the "commandTopic" parameter within the differential drive plugin.

Any suggestions for identifying the correct cmd_vel topic?

zzuxzt, 2024-02-18:
First, you need to know the control command topic for your robot, e.g. it is '/cmd_vel'. Then, the simplest way is that you can remap the 'cmd_vel' topic from DRL-VO to your control command topic in the launch file [drl_vo_inference.launch](https://github.com/TempleRAIL/drl_vo_nav/blob/baa970c30cabe94a8b0eb81bb0ad0b97261f2458/drl_vo/launch/drl_vo_inference.launch#L13):
```
<!-- Mix cmd vel publisher -->
  <node name="mix_cmd_vel" pkg="drl_vo_nav" type="cmd_vel_pub.py" output="screen" >
    <remap from="cmd_vel" to="/cmd_vel"/>
  </node>  
```

zzuxzt, 2024-02-21:
If you can control your robot movement using control commands from the trained model, then in theory replacing the differential driver would not cause too many problems. Just make sure you use the correct control command topic to control the robot's movement.

FurkanEdizkan, 2024-02-29:
I still have problems with controller, I really don't know why I can't make robot move with [libgazebo_ros_diff_drive.so](https://classic.gazebosim.org/tutorials?tut=ros_gzplugins#DifferentialDrive) and [kobuki_controller](https://github.com/yujinrobot/kobuki_description/blob/master/urdf/kobuki_gazebo.urdf.xacro) doesn't make robot move either. *cmd_vel* topics are published correctly and they are remappted to correct topics even all velocity topics working and gazebo listens to correct *cmd_vel* topic but no movement can be observed eventhough *cmd_vel* is non zero. Only with [libgazebo_ros_planar_move.so](https://classic.gazebosim.org/tutorials?tut=ros_gzplugins#PlanarMovePlugin) make robot move and it has it's own problems :sweat:.

But I don't think there are any other thing related to DRL-VO which I couldn't understand. All my problems are about controller it seems I will try to fix it and post the solution so anyone who faces same issue can solve it however till then I will close this issue

Thank you

closed 2024-02-29 by FurkanEdizkan