TempleRAIL/drl_vo_nav #22: what are they ( mobile_base and world tf)

CAI23sbP, 2024-02-26:
Hi @zzuxzt @philipdames  How are you?
Anyway, i have two questions about this code.

1. As you know, [in this code](https://github.com/TempleRAIL/drl_vo_nav/blob/drl_vo/drl_vo/src/track_ped_pub.py) uses "mobile_base" and "world" TF.  What are they?. i can't see them in ros_tf_tree.  Could you explain them?

2. When i have launched a training, Odom TF was changing Whenever episode (or scenario was changed).
How to fix it until training is ended? i want to fix it and to merge with map TF.

zzuxzt, 2024-02-27:
1. The "mobile_base" and "world" are just the names of the models in the Gazebo world and do not belong to the TF. I just used them to get the robot model state in the Gazebo world through a service provided by the Gazebo simulator. More information about it can be found at [Get Model State Example](https://classic.gazebosim.org/tutorials?tut=ros_comm#GetModelStateExample)
2. The changes of the odom TF are because the simulated world/scenario has been reset, which means the robot has restarted at a random position and its odom TF should change. It does not affect the training and the localization is provided by the [amcl](http://wiki.ros.org/amcl) package. 

​

CAI23sbP, 2024-02-28:
@zzuxzt  Hi , Thanks to your reply.
I have another single question.
A demo model weight which is called drl_vo.zip, How long it takes to train?.
I mean that number of steps are spended?
It is not enough to test?

zzuxzt, 2024-02-29:
When training from the pre-trained model, it takes about 1M steps.

closed 2024-02-29 by CAI23sbP