TempleRAIL/drl_vo_nav #23: run "bash run_drl_vo_policy_training_desktop.sh ./drl_vo_runs "  meet these error

luyao2001, 2024-04-19:
started roslaunch server http://robot-ZHENGJIUZHE-REN7000K-28IMB:38663/

SUMMARY
========

PARAMETERS
 * /amcl/base_frame_id: base_footprint
 * /amcl/global_frame_id: map
 * /amcl/gui_publish_rate: 10.0
 * /amcl/initial_pose_a: 0.13
 * /amcl/initial_pose_x: 1.0
 * /amcl/initial_pose_y: 0.0
 * /amcl/kld_err: 0.05
 * /amcl/kld_z: 0.99
 * /amcl/laser_lambda_short: 0.1
 * /amcl/laser_likelihood_max_dist: 2.0
 * /amcl/laser_max_beams: 60
 * /amcl/laser_max_range: 12.0
 * /amcl/laser_model_type: likelihood_field
 * /amcl/laser_sigma_hit: 0.2
 * /amcl/laser_z_hit: 0.5
 * /amcl/laser_z_max: 0.05
 * /amcl/laser_z_rand: 0.5
 * /amcl/laser_z_short: 0.05
 * /amcl/max_particles: 2000
 * /amcl/min_particles: 500
 * /amcl/odom_alpha1: 0.2
 * /amcl/odom_alpha2: 0.2
 * /amcl/odom_alpha3: 0.2
 * /amcl/odom_alpha4: 0.2
 * /amcl/odom_alpha5: 0.1
 * /amcl/odom_frame_id: odom
 * /amcl/odom_model_type: diff
 * /amcl/recovery_alpha_fast: 0.0
 * /amcl/recovery_alpha_slow: 0.0
 * /amcl/resample_interval: 1
 * /amcl/transform_tolerance: 3.5
 * /amcl/update_min_a: 0.2
 * /amcl/update_min_d: 0.25
 * /amcl/use_map_topic: False
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /cmd_vel_mux/yaml_cfg_file: /data1/crowllt_ws...
 * /drl_vo_cmd/log_dir: ./drl_vo_runs
 * /drl_vo_cmd/model_file: /data1/crowllt_ws...
 * /gazebo/enable_ros_network: True
 * /move_base/DWAPlannerROS/acc_lim_theta: 2.0
 * /move_base/DWAPlannerROS/acc_lim_x: 1.0
 * /move_base/DWAPlannerROS/acc_lim_y: 1.0
 * /move_base/DWAPlannerROS/forward_point_distance: 0.5
 * /move_base/DWAPlannerROS/global_frame_id: odom
 * /move_base/DWAPlannerROS/goal_distance_bias: 24.0
 * /move_base/DWAPlannerROS/max_scaling_factor: 0.2
 * /move_base/DWAPlannerROS/max_vel_rot: 3.0
 * /move_base/DWAPlannerROS/max_vel_trans: 0.5
 * /move_base/DWAPlannerROS/max_vel_x: 0.5
 * /move_base/DWAPlannerROS/max_vel_y: 0.0
 * /move_base/DWAPlannerROS/min_vel_rot: 0.4
 * /move_base/DWAPlannerROS/min_vel_trans: 0.1
 * /move_base/DWAPlannerROS/min_vel_x: 0.0
 * /move_base/DWAPlannerROS/min_vel_y: 0.0
 * /move_base/DWAPlannerROS/occdist_scale: 0.5
 * /move_base/DWAPlannerROS/oscillation_reset_dist: 0.05
 * /move_base/DWAPlannerROS/path_distance_bias: 64.0
 * /move_base/DWAPlannerROS/publish_cost_grid_pc: True
 * /move_base/DWAPlannerROS/publish_traj_pc: True
 * /move_base/DWAPlannerROS/scaling_speed: 0.25
 * /move_base/DWAPlannerROS/sim_time: 1.0
 * /move_base/DWAPlannerROS/stop_time_buffer: 0.2
 * /move_base/DWAPlannerROS/theta_stopped_vel: 0.4
 * /move_base/DWAPlannerROS/trans_stopped_vel: 0.1
[... 476 lines cut ...]
  File "/home/robot/anaconda3/envs/torch211/lib/python3.8/site-packages/gym/envs/registration.py", line 105, in make
    env = spec.make(**kwargs)
  File "/home/robot/anaconda3/envs/torch211/lib/python3.8/site-packages/gym/envs/registration.py", line 75, in make
    env = cls(**_kwargs)
  File "/data1/crowllt_ws/src/drl_vo_nav/drl_vo/src/turtlebot_gym/turtlebot_gym/envs/drl_nav_env.py", line 136, in __init__
    self.gazebo.pauseSim()   
  File "/data1/crowllt_ws/src/drl_vo_nav/drl_vo/src/turtlebot_gym/turtlebot_gym/envs/gazebo_connection.py", line 41, in pauseSim
    rospy.wait_for_service('/gazebo/pause_physics')
[robot_state_publisher-10] killing on exit
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 166, in wait_for_service
    raise ROSInterruptException("rospy shutdown")
rospy.exceptions.ROSInterruptException: rospy shutdown
[bumper2pointcloud-9] killing on exit
[cmd_vel_mux-8] killing on exit
[mobile_base_nodelet_manager-7] killing on exit
[gazebo_map_broadcaster-5] killing on exit
[gazebo_gui-3] killing on exit
[gazebo-2] killing on exit
terminate called without an active exception
Aborted (core dumped)
[gazebo_gui-3] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

luyao2001, 2024-04-19:
[WARN] [1713518138.974658, 0.134000]: Could not get robot pose
[WARN] [1713518139.009384, 0.167000]: Could not get robot pose
[INFO] [1713518139.014711, 0.171000]: Spawning model: actor_id = 1
[WARN] [1713518139.055238, 0.209000]: Could not get robot pose
[WARN] [1713518139.081983, 0.234000]: Could not get robot pose
[WARN] [1713518139.116563, 0.267000]: Could not get robot pose
[WARN] [1713518139.150900, 0.300000]: Could not get robot pose

[move_base-17] process has died [pid 520664, exit code 127, cmd /opt/ros/noetic/lib/move_base/move_base cmd_vel:=/move_base/cmd_vel odom:=odom scan:=scan __name:=move_base __log:=/home/robot/.ros/log/6020d574-fe2d-11ee-92c5-3102a54b67ec/move_base-17.log].
log file: /home/robot/.ros/log/6020d574-fe2d-11ee-92c5-3102a54b67ec/move_base-17*.log
[INFO] [1713518144.443544, 4.089000]: Spawning model: actor_id = 23
[INFO] [1713518144.684117, 4.267000]: Spawning model: actor_id = 24
[FATAL] [1713518144.717703, 4.277000]: Sensor topic "/move_base/status" is not available. Waiting...

zzuxzt, 2024-04-20:
Sorry, I found no obvious errors. I just noticed that you are using conda with ros, which could cause some potential problems if you don't handle library dependencies well. By the way, 'Could not get robot pose' is just a warning log, please wait a while for the system to start working. 

If it still does not work, can you describe it in more detail?

luyao2001, 2024-04-21:
[image]
[image]
[image]
[image]

luyao2001, 2024-04-21:
I am very glad to receive your reply. The above pictures show the ERROR and Warning messages I encounter when I run "roslaunch drl_vo_nav drl_vo_nav_train.launch log_dir:=./drl_vo_runs".

Reference some details in #9 
I also try to run "roslaunch turtlebot_gazebo turtlebot_world.launch" and "roslaunch turtlebot_gazebo amcl_demo.launch" to test turtlebot simulator. I meet the following error when run commend "roslaunch turtlebot_gazebo amcl_demo.launch". 
[image]
[image]

zzuxzt, 2024-04-21:
It looks like there are some issues on your move_base package, please try to use the following commands to reinstall navigation packages:
```
sudo apt-get install ros-noetic-move-base*
sudo apt-get install ros-noetic-map-server*
sudo apt-get install ros-noetic-amcl*
sudo apt-get install ros-noetic-navigation*
```

luyao2001, 2024-04-22:
> It looks like there are some issues on your move_base package, please try to use the following commands to reinstall navigation packages:
>
> ```
> sudo apt-get install ros-noetic-move-base*
> sudo apt-get [...]

I have run the above commands, but it is not useful for my error. I still meet this error.
[image]

luyao2001, 2024-04-22:
I also try to reinstall ros-noetic, but I still meet this error.

zzuxzt, 2024-04-22:
It should be the problem with your ros environment, as you can see the 'move_base symbol lookup error' in your log. This error is not related the DRL-VO package. I would suggest you uninstall all ros packages, clean the ros environment, and reinstall all ros packages again (or test on a new computer with the clean environment). 

After reinstalling required ros packages, double-check if there are any issues with the move_base navigation by running the following command again:
```
roslaunch turtlebot_gazebo turtlebot_world.launch
roslaunch turtlebot_gazebo amcl_demo.launch
roslaunch turtlebot_rviz_launchers view_navigation.launch
```
If it still does not work, consider git clone move_base package's source code and compile it in your catkin_ws workspace. Or you can just directly use my prebuilt singularity container.

luyao2001, 2024-04-23:
I have solved this issue by a lot of effort. This error is caused by "liblayers.so", which lack the dependency named "voxel_grid::VoxelGrid::VoxelGrid". The source file is in
[https://github.com/ros-planning/navigation/blob/noetic-devel/costmap_2d/CMakeLists.txt](https://github.com/ros-planning/navigation/blob/noetic-devel/costmap_2d/CMakeLists.txt)

My solution is following:
[image]
Add this line "../voxel_grid/src/voxel_grid.cpp"

closed 2024-04-24 by luyao2001