TempleRAIL/drl_vo_nav #26: ‘QObject’ is an inaccessible base of ‘rviz::_AdditionalTopicSubscriber’

yriuss, 2024-05-12:
Hi, when I use the command ```catkin_make```, after following all the instructions I get the following error:

```
everything seems ok until here
....
[ 97%] Built target kobuki_auto_docking_nodelet
[ 97%] Built target kobuki_nodelet
In file included from /opt/ros/noetic/include/rviz/message_filter_display.h:40,
                 from /root/catkin_ws/src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/person_display_common.h:40,
                 from /root/catkin_ws/src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/social_activities_display.h:39,
                 from /root/catkin_ws/src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/social_activities_display.cpp:36:
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h: In instantiation of ‘rviz::RosTopicProperty::RosTopicProperty(const QString&, const QString&, const QString&, const QString&, rviz::Property*, Func&&, const R*) [with Func = const char (&)[15]; R = rviz::_AdditionalTopicSubscriber]’:
/root/catkin_ws/src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h:71:125:   required from here
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h:59:5: error: ‘QObject’ is an inaccessible base of ‘rviz::_AdditionalTopicSubscriber’
   59 |     connect(receiver, std::forward<Func>(changed_slot));
      |     ^~~~~~~
In file included from /root/catkin_ws/src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h:52,
                 from /root/catkin_ws/src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/tracked_persons_cache.h:40,
                 from /root/catkin_ws/src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/tracked_persons_cache.cpp:31:
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h: In instantiation of ‘rviz::RosTopicProperty::RosTopicProperty(const QString&, const QString&, const QString&, const QString&, rviz::Property*, Func&&, const R*) [with Func = const char (&)[15]; R = rviz::_AdditionalTopicSubscriber]’:
/root/catkin_ws/src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h:71:125:   required from here
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h:59:5: error: ‘QObject’ is an inaccessible base of ‘rviz::_AdditionalTopicSubscriber’
   59 |     connect(receiver, std::forward<Func>(changed_slot));
      |     ^~~~~~~
In file included from /root/catkin_ws/build/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/../../../../../src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h:52,
                 from /root/catkin_ws/build/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/moc_additional_topic_subscriber.cpp:9:
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h: In instantiation of ‘rviz::RosTopicProperty::RosTopicProperty(const QString&, const QString&, const QString&, const QString&, rviz::Property*, Func&&, const R*) [with Func = const char (&)[15]; R = rviz::_AdditionalTopicSubscriber]’:
/root/catkin_ws/build/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/../../../../../src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h:71:125:   required from here
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h:59:5: error: ‘QObject’ is an inaccessible base of ‘rviz::_AdditionalTopicSubscriber’
   59 |     connect(receiver, std::forward<Func>(changed_slot));
      |     ^~~~~~~
In file included from /opt/ros/noetic/include/rviz/message_filter_display.h:40,
                 from /root/catkin_ws/build/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/../../../../../src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/person_display_common.h:40,
                 from /root/catkin_ws/build/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/../../../../../src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/social_activities_display.h:39,
                 from /root/catkin_ws/build/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/moc_social_activities_display.cpp:9:
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h: In instantiation of ‘rviz::RosTopicProperty::RosTopicProperty(const QString&, const QString&, const QString&, const QString&, rviz::Property*, Func&&, const R*) [with Func = const char (&)[15]; R = rviz::_AdditionalTopicSubscriber]’:
/root/catkin_ws/build/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/../../../../../src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h:71:125:   required from here
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h:59:5: error: ‘QObject’ is an inaccessible base of ‘rviz::_AdditionalTopicSubscriber’
   59 |     connect(receiver, std::forward<Func>(changed_slot));
      |     ^~~~~~~
In file included from /opt/ros/noetic/include/rviz/message_filter_display.h:40,
                 from /root/catkin_ws/src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/person_display_common.h:40,
                 from /root/catkin_ws/src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/tracked_groups_display.h:40,
                 from /root/catkin_ws/src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/tracked_groups_display.cpp:35:
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h: In instantiation of ‘rviz::RosTopicProperty::RosTopicProperty(const QString&, const QString&, const QString&, const QString&, rviz::Property*, Func&&, const R*) [with Func = const char (&)[15]; R = rviz::_AdditionalTopicSubscriber]’:
/root/catkin_ws/src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h:71:125:   required from here
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h:59:5: error: ‘QObject’ is an inaccessible base of ‘rviz::_AdditionalTopicSubscriber’
   59 |     connect(receiver, std::forward<Func>(changed_slot));
      |     ^~~~~~~
In file included from /opt/ros/noetic/include/rviz/message_filter_display.h:40,
                 from /root/catkin_ws/build/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/../../../../../src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/person_display_common.h:40,
                 from /root/catkin_ws/build/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/../../../../../src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/social_relations_display.h:36,
                 from /root/catkin_ws/build/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/moc_social_relations_display.cpp:9:
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h: In instantiation of ‘rviz::RosTopicProperty::RosTopicProperty(const QString&, const QString&, const QString&, const QString&, rviz::Property*, Func&&, const R*) [with Func = const char (&)[15]; R = rviz::_AdditionalTopicSubscriber]’:
/root/catkin_ws/build/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/../../../../../src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/additional_topic_subscriber.h:71:125:   required from here
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h:59:5: error: ‘QObject’ is an inaccessible base of ‘rviz::_AdditionalTopicSubscriber’
   59 |     connect(receiver, std::forward<Func>(changed_slot));
      |     ^~~~~~~
In file included from /opt/ros/noetic/include/rviz/message_filter_display.h:40,
                 from /root/catkin_ws/build/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/../../../../../src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/person_display_common.h:40,
                 from /root/catkin_ws/build/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/../../../../../src/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/tracked_groups_display.h:40,
                 from /root/catkin_ws/build/pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/src/moc_tracked_groups_display.cpp:9:
/opt/ros/noetic/include/rviz/properties/ros_topic_property.h: In instantiation of ‘rviz::RosTopicProperty::RosTopicProperty(const QString&, const QString&, const QString&, const QString&, rviz::Property [... cut ...]
make[2]: *** [pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/CMakeFiles/spencer_tracking_rviz_plugin.dir/build.make:160: pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/CMakeFiles/spencer_tracking_rviz_plugin.dir/src/human_attributes_display.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:26892: pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin/CMakeFiles/spencer_tracking_rviz_plugin.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j12 -l12" failed

```

zzuxzt, 2024-05-15:
It seems to be a compatibility issue between Rviz and QT. Please check the versions of them. If it cannot be solved, you can directly remove the 'pedsim_ros_with_gazebo/2ndparty/spencer_tracking_rviz_plugin' package. It only provides the pedestrian Rviz visulaization and does not affect the robot navigation and other things.

closed 2024-05-25 by yriuss

FurkanEdizkan, 2024-05-29:
Solution -> https://github.com/ai-winter/ros_motion_planning/issues/72

@yriuss