TempleRAIL/drl_vo_nav #27: About the Lidar Data preprocessing

lehoangan2906, 2024-06-21:
Hi, I want to thank you and your team for your hard work in this amazing project. I'm currently reviewing your paper and code. I'm not quite understand how to get 1x80x80 dimensional lidar map from 20x720 Min & Avg pooled data from the paper. When I read your code to understand more, I'm struggling to find these preprocessing functions. Can you explain this further to me, and if convenient, can you guide me through your code structure (about how things are organized)? Again, I am very appreciate your time and effort you put into this project!

zzuxzt, 2024-06-26:
Thank you for your interest in our work. Sorry, I have been busy with some personal issues these days. 

The data preprocessing code is in the [cnn_data_pub.py](https://github.com/TempleRAIL/drl_vo_nav/blob/drl_vo/drl_vo/src/cnn_data_pub.py). The network structure code is in the [custom_cnn_full.py](https://github.com/TempleRAIL/drl_vo_nav/blob/drl_vo/drl_vo/src/custom_cnn_full.py). The training code is in the [drl_vo_train.py](https://github.com/TempleRAIL/drl_vo_nav/blob/drl_vo/drl_vo/src/drl_vo_train.py). The navigation code is in the [drl_vo_inference.py](https://github.com/TempleRAIL/drl_vo_nav/blob/drl_vo/drl_vo/src/drl_vo_inference.py). 

Hopefully these simple descriptions can give you better guidance.  I will restructure the code later when I am less busy.

lehoangan2906, 2024-06-26:
Hi zzuxzt, thank you for your prompt reply. Do you mind if I ask you another question?

When I set up the turtlebot2 as your README.md, I encountered the following issue:

"The following packages have unmet dependencies: ros-noetic-joystick-drivers : Depends: ros-noetic-ps3joy but it is not installable Depends: ros-noetic-wiimote but it is not installable E: Unable to correct problems, you have held broken packages."

I have been searching for the solutions for several days but still don't know how to resolve it. Do you have any advice? 

I'm very appreciate your guidance with the code structure and I will analyze them further soon. Wish you do well with your work and I'm looking forward for your reply 😁

zzuxzt, 2024-06-26:
You can try to remove the ros-noetic-joystick-drivers package since the DRL-VO did not use it. I guess the reason is that the packages ps3joy and wiimote cannot support Noetic as mentioned in [here](https://github.com/ros-drivers/joystick_drivers/issues/204).

lehoangan2906, 2024-06-27:
I'm very appreciate your help. Can I contact to ask you more if I encounter more problem?

zzuxzt, 2024-06-27:
Feel free to post your questions. I will answer your questions when I am free.

lehoangan2906, 2024-07-03:
Hi @zzuxzt, I'm currently reading the Simulation configuration and Hardware configuration sections in your paper. I want to ask you that why didn't you both train the DRL network and run the simulations on the DGX-1 server, but split into server specific task and desktop specific task? And if there are someway to train the model without the need to activate Rviz or Gazebo, can you please guide me?

Another point I noticed is that between the training/simulation phase and the real-life hardware testing phase, you used different Ubuntu and ROS versions. Is that optional or is there any specific reason for you to do so? Because I assume that there will be conflicts between the versions.

Again, thank you for your time and kindness. I hope to get your reply soon!
[image]
[image]

lehoangan2906, 2024-07-03:
Beside, I think you should update the turtlebot2 installation file as some of the packages are no longer supported: https://raw.githubusercontent.com/zzuxzt/turtlebot2_noetic_packages/master/turtlebot2_noetic_install.sh 😁

zzuxzt, 2024-07-04:
> Hi @zzuxzt, I'm currently reading the Simulation configuration and Hardware configuration sections in your paper. I want to ask you that why didn't you both train the DRL network and run the simulations on the DGX-1 server, but split into server specific task and desktop specific task? And if there are someway to train the model without the need to activate Rviz or Gazebo, can you please guide me?
> 
> Another point I noticed is that between the training/simulation phase and the real-life hardware testing phase, you used different Ubuntu and ROS versions. Is that optional or is there any specific reason for you to do so? Because I assume that there will be conflicts between the versions.
> 
> Again, thank you for your time and kindness. I hope to get your reply soon! ![Screenshot 2024-07-03 at 9 01 57 AM](https://private-user-images.githubusercontent.com/75709930/345256451-7ae3db8d-e55d-471c-8910-4c173c20adb0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAxMDQ3OTIsIm5iZiI6MTcyMDEwNDQ5MiwicGF0aCI6Ii83NTcwOTkzMC8zNDUyNTY0NTEtN2FlM2RiOGQtZTU1ZC00NzFjLTg5MTAtNGMxNzNjMjBhZGIwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzA0VDE0NDgxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlhN2ZjYjc4Yzg1OGFhNWFkM2Y0NjJlM2ZlMTgzNTViYWE1OGU0M2Y5NGRjNWY3NDhmNGUxZjk1NGM1NTBjNWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.nIraZifoeNvZrJU_zbJ0lC-l8js6XamgFuLWD_N3mKE) ![Screenshot 2024-07-03 at 9 02 21 AM](https://private-user-images.githubusercontent.com/75709930/345256515-8f29cacb-f6e2-4d8f-85aa-605a57c4cf6b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAxMDQ3OTIsIm5iZiI6MTcyMDEwNDQ5MiwicGF0aCI6Ii83NTcwOTkzMC8zNDUyNTY1MTUtOGYyOWNhY2ItZjZlMi00ZDhmLTg1YWEtNjA1YTU3YzRjZjZiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzA0VDE0NDgxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRiYWFlZDhlZjZiMWYzODc0ZGIwYWJiYzllNmRiMDAwODBiYThhMzRiYjVhNGMxMmZlYTVmZWI1YjZjYjIyNWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Iz4ZGTQtIXP_boBEtraLu-xI2_E0w1SuRoOoiMqjqcg)

1. The direct reason is that the DGX-1 server does not support a GUI, which makes it difficult to run simulations. More importantly, if your algorithm only works well on the same machine it was trained on, then your algorithm has limited generalization capabilities and is not a good algorithm. Computer machines should not be a factor influencing algorithms.

2. You can easily find many other python-based simulators without using the Gazebo or rviz. For example, you can find many interesting works that do not use Gazebo in the second part of my paper, "Related Works".

3. The difference in versions is due to the fact that the robot computer TX2 only supports Ubuntu 18.04. This is also a tricky thing in hardware experiments, where you need to deal with compatibility issues between different versions.

zzuxzt, 2024-07-04:
> Beside, I think you should update the turtlebot2 installation file as some of the packages are no longer supported: https://raw.githubusercontent.com/zzuxzt/turtlebot2_noetic_packages/master/turtlebot2_noetic_install.sh 😁

Thanks for your reminder. I will find time to update it since it was created a few years ago and many packages have changed.

lehoangan2906, 2024-07-05:
I appreciate your help!

closed 2024-07-20 by zzuxzt