TempleRAIL/drl_vo_nav #28: Training and implementing an other different robot instead of turtlebot2

tendflow, 2024-07-16:
Hello, thank you very much for uploading the code. I have used your project to train a model. Now I want to test your project on a real robot (not turtlebot2) in gazebo environment and real world. What do I need to do for this? Also, if I want to train a project similar to yours, is the dataset you uploaded available? I would be grateful if you answer my question at your free time.

zzuxzt, 2024-07-20:
Thank you for your interest in our work. I'm sorry for being late as I was busy with my personal affairs in these days.

1. To deploy our DRL-VO policy to the other robots, you need to make sure the input data structure in the [cnn_data_pub.py](https://github.com/TempleRAIL/drl_vo_nav/blob/drl_vo/drl_vo/src/cnn_data_pub.py) is correct (e.g., downsampling the lidar scans to 80 points and setting the pedestrian tracking topic to your own). But I will update a new branch for deployment and as a baseline in the next few days when I am less busy.

2. We are using deep reinforcement learning to train our DRL-VO, which does not require a dataset but requires experience in the interaction between the robot and the environment. If you want to train a new policy, you can directly use our current project and simulator and change the network or other reward settings to train it.

tendflow, 2024-07-21:
OK,I got it. Thanks for your help.

tendflow, 2024-08-19:
Hello, sorry about bothering you again. If I want to use my simulator, network and reward settings to train a new policy, is your experience in this project still available?

zzuxzt, 2024-08-24:
Yes, because the lessons learned from our work are universal and should be applicable in different training simulators.

tendflow, 2024-08-24:
OK, thanks for your help.

closed 2024-08-24 by tendflow

CAI23sbP, 2025-02-02:
@zzuxzt  Hi there, I want to use VO-novel reward for my robot. 
In my case, the robot radius is 0.6m. How to modified your code in VO-novel reward? 

Here is your code. 
`                    if(ped_dis <= 7):
                        ped_theta = np.arctan2(p_y, p_x)
                        vo_theta = np.arctan2(3*self.robot_radius, np.sqrt(ped_dis**2 - (3*self.robot_radius)**2))
`

When i run this i have a single warning issue, see under code 
 
`rewards/drl_vo_reward.py:113: RuntimeWarning: invalid value encountered in sqrt
  vo_theta = np.arctan2(3*self.robot_radius, np.sqrt(ped_dis**2 - (3*self.robot_radius)**2))
`