TempleRAIL/drl_vo_nav #29: How to test

Shiqi-7-7, 2024-08-12:
Hello, thank you very much for your work. I have read your article and tried to reproduce code.How to test this model?
[image]
How do I obtain the curve graph during the training process and the charts(success rate,time,length,speed) in your paper?
Thank you very much!

zzuxzt, 2024-08-24:
Thanks for your interest in our work. If you want to test the trained model, you can deploy the control policy on the robot, publish a predefined series of goal points,  and let the robot navigate through a predefined series of goal points around the corresponding test environment. For example, you can publish a set of goal points to test the policy using the following command:
```
roslaunch drl_vo_nav publish_goal_sequence.launch
```

You can view the training curve using tensorboard using the following command:
```
tensorboard --logdir="./runs/"
```

GGHaozhu, 2024-10-14:
Hello, thank you very much for your work.When I was testing roslaunch drl_vo_nav publish_goal_sequence.launch, when the robot reached the first point, it would report an error: Received comm state ACTIVE when in simple state DONE with SimpleActionClient in NS /move_base, when the second point is reached, an error message is displayed: SimpleActionClient received DONE twice, and then the robot doesn't move. What's the reason for that? Is there a solution?

zzuxzt, 2024-10-17:
It seems that the SimpleActionClient thinks the goal has been completed, but it receives another status update from the action server that the goal is still active. Did you publish a goal using the Rviz GUI while still roslaunch publish_goal_sequence? In addition, what were all your commands for testing the robot? Did you use the Lobby World to test? I suspect you didn't launch everything correctly.

GGHaozhu, 2024-10-18:
> It seems that the SimpleActionClient thinks the goal has been completed, but it receives another status update from the action server that the goal is still active. Did you publish a goal using the [...]

Thanks for your reply, I used the autolab world for testing, using your source publish_goal_sequence.py file. I first started the test node sh run_drl_vo_navigation_demo. Then launch publish_goal_sequence_for_autolab.launch. Thank you for helping me troubleshoot the problem，among
<launch>
<arg name="scene_file" default="$(find pedsim_simulator)/scenarios/autolab/autolab_25.xml"/>
<arg name="world_name" default="$(find pedsim_gazebo_plugin)/worlds/autolab.world"/>
<arg name="gui" default="true" doc="Bring up the Gazebo graphical interface"/>
<arg name="pose_initial_x" default="1.0"/>
<arg name="pose_initial_y" default="8.0"/>
<arg name="map_file" default="$(find robot_gazebo)/maps/gazebo_autolab/gazebo_autolab.yaml"/>
<arg name="initial_pose_x" default="0.0"/>
<arg name="initial_pose_y" default="0.0"/>
<arg name="initial_pose_a" default="0.0"/>
<arg name="model_file" default="$(find drl_vo_nav)/src/model/drl_vo.zip"/>
<! -- Output -->
<arg name="output" default="log"/>
<arg name="enable_opencv" default="true"/>
<arg name="enable_console_output" default="true"/>
<arg name="rviz" default="true"/><! -- Pedsim Gazebo -->
<include file="$(find pedsim_simulator)/launch/robot.launch">
<arg name="scene_file" value="$(arg scene_file)"/>
<arg name="world_name" value="$(arg world_name)"/>
<arg name="gui" value="$(arg gui)"/>
<arg name="pose_initial_x" value="$(arg pose_initial_x)"/>
<arg name="pose_initial_y" value="$(arg pose_initial_y)"/>
</include><! -- AMCL -->
<include file="$(find robot_gazebo)/launch/amcl_demo_drl.launch">
<arg name="map_file" value="$(arg map_file)"/>
<arg name="initial_pose_x" value="$(arg initial_pose_x)"/>
<arg name="initial_pose_y" value="$(arg initial_pose_y)"/>
<arg name="initial_pose_a" value="$(arg initial_pose_a)"/>
</include><! -- CNN DATA -->
<include file="$(find drl_vo_nav)/launch/nav_cnn_data.launch"/><! -- DRL-VO Control Policy -->
<include file="$(find drl_vo_nav)/launch/drl_vo_inference.launch">
<arg name="model_file" value="$(arg model_file)"/>
</include><! -- Rviz -->
<include file="$(find robot_gazebo)/launch/view_navigation.launch" if="$(arg rviz)"/>

By the way, I would also like to ask you, if you want to do the physical test, do you need to prepare the list of objects only need to buy turtlebot2? Main control configuration:
i7 8g main control pc, hardware configuration: chassis + support + large battery; Visual configuration Kinect V2, is this configuration OK?

GGHaozhu, 2024-10-18:
> It seems that the SimpleActionClient thinks the goal has been completed, but it receives another status update from the action server that the goal is still active.  Did you publish a goal using the [...]

Thanks for your reply, I used the autolab world for testing, using your source publish_goal_sequence.py file.  I first started the test node sh run_drl_vo_navigation_demo.  Then launch publish_goal_sequence_for_autolab.launch.  Thank you for helping me troubleshoot the problem，among
<launch>
<arg name="scene_file" default="$(find pedsim_simulator)/scenarios/autolab/autolab_25.xml"/>
<arg name="world_name" default="$(find pedsim_gazebo_plugin)/worlds/autolab.world"/>
<arg name="gui" default="true" doc="Bring up the Gazebo graphical interface"/>
<arg name="pose_initial_x" default="1.0"/>
<arg name="pose_initial_y" default="8.0"/>
<arg name="map_file" default="$(find robot_gazebo)/maps/gazebo_autolab/gazebo_autolab.yaml"/>
<arg name="initial_pose_x" default="0.0"/>
<arg name="initial_pose_y" default="0.0"/>
<arg name="initial_pose_a" default="0.0"/>
<arg name="model_file" default="$(find drl_vo_nav)/src/model/drl_vo.zip"/>
<!  -- Output -->
<arg name="output" default="log"/>
<arg name="enable_opencv" default="true"/>
<arg name="enable_console_output" default="true"/>
<arg name="rviz" default="true"/><!  -- Pedsim Gazebo -->
<include file="$(find pedsim_simulator)/launch/robot.launch">
<arg name="scene_file" value="$(arg scene_file)" />
<arg name="world_name" value="$(arg world_name)" />
<arg name="gui" value="$(arg gui)" />
<arg name="pose_initial_x" value="$(arg pose_initial_x)" />
<arg name="pose_initial_y" value="$(arg pose_initial_y)" />
</include><!  -- AMCL -->
<include file="$(find robot_gazebo)/launch/amcl_demo_drl.launch">
<arg name="map_file" value="$(arg map_file)" />
<arg name="initial_pose_x" value="$(arg initial_pose_x)" />
<arg name="initial_pose_y" value="$(arg initial_pose_y)" />
<arg name="initial_pose_a" value="$(arg initial_pose_a)" />
</include><!  -- CNN DATA -->
<include file="$(find drl_vo_nav)/launch/nav_cnn_data.launch"/><!  -- DRL-VO Control Policy -->
<include file="$(find drl_vo_nav)/launch/drl_vo_inference.launch">
<arg name="model_file" value="$(arg model_file)" />
</include><!  -- Rviz -->
<include file="$(find robot_gazebo)/launch/view_navigation. launch" if="$(arg rviz) "/>

By the way, I would also like to ask you, if you want to do the physical test, do you need to prepare the list of objects only need to buy turtlebot2?  Main control configuration:
i7 8g main control pc, hardware configuration: chassis + support + large battery;  Visual configuration Kinect V2, is this configuration OK?

GGHaozhu, 2024-10-18:
> It seems that the SimpleActionClient thinks the goal has been completed, but it receives another status update from the action server that the goal is still active.  Did you publish a goal using the [...]

Thanks for your reply, I used the autolab world for testing, using your source publish_goal_sequence.py file.  I first started the test node sh run_drl_vo_navigation_demo.  Then launch publish_goal_sequence_for_autolab.launch.  Thank you for helping me troubleshoot the problem，among`<launch>
<arg name="scene_file" default="$(find pedsim_simulator)/scenarios/autolab/autolab_25.xml"/>
<arg name="world_name" default="$(find pedsim_gazebo_plugin)/worlds/autolab.world"/>
<arg name="gui" default="true" doc="Bring up the Gazebo graphical interface"/>
<arg name="pose_initial_x" default="1.0"/>
<arg name="pose_initial_y" default="8.0"/>
<arg name="map_file" default="$(find robot_gazebo)/maps/gazebo_autolab/gazebo_autolab.yaml"/>
<arg name="initial_pose_x" default="0.0"/>
<arg name="initial_pose_y" default="0.0"/>
<arg name="initial_pose_a" default="0.0"/>
<arg name="model_file" default="$(find drl_vo_nav)/src/model/drl_vo.zip"/>
<!  -- Output -->
<arg name="output" default="log"/>
<arg name="enable_opencv" default="true"/>
<arg name="enable_console_output" default="true"/>
<arg name="rviz" default="true"/><!  -- Pedsim Gazebo -->
<include file="$(find pedsim_simulator)/launch/robot.launch">
<arg name="scene_file" value="$(arg scene_file)" />
<arg name="world_name" value="$(arg world_name)" />
<arg name="gui" value="$(arg gui)" />
<arg name="pose_initial_x" value="$(arg pose_initial_x)" />
<arg name="pose_initial_y" value="$(arg pose_initial_y)" />
</include><!  -- AMCL -->
<include file="$(find robot_gazebo)/launch/amcl_demo_drl.launch">
<arg name="map_file" value="$(arg map_file)" />
<arg name="initial_pose_x" value="$(arg initial_pose_x)" />
<arg name="initial_pose_y" value="$(arg initial_pose_y)" />
<arg name="initial_pose_a" value="$(arg initial_pose_a)" />
</include><!  -- CNN DATA -->
<include file="$(find drl_vo_nav)/launch/nav_cnn_data.launch"/><!  -- DRL-VO Control Policy -->
<include file="$(find drl_vo_nav)/launch/drl_vo_inference.launch">
<arg name="model_file" value="$(arg model_file)" />
</include><!  -- Rviz -->
<include file="$(find robot_gazebo)/launch/view_navigation. launch" if="$(arg rviz) "/>
`
By the way, I would also like to ask you, if you want to do the physical test, do you need to prepare the list of objects only need to buy turtlebot2?  Main control configuration:
i7 8g main control pc, hardware configuration: chassis + support + large battery;  Visual configuration Kinect V2, is this configuration OK?

zzuxzt, 2024-10-20:
I don't see any problem with your process. Were you able to send goals using the Rviz tool? If yes, did you install the package in your Ubuntu? I suspect there are some issues with your ROS environment. Did you try to use the singularity container? It would be good if you could share a debug video. 

You can use your own robot and do not need to buy a Turtlebot 2 robot. Just make true your robot can provide pedestrian tracking information and lidar information. Your configuration sounds good, but it would be even better if you had a GPU. I will update the deployment code with a Jackal robot as an example soon.

GGHaozhu, 2024-10-22:
Thanks for your reply, I am not using singularity containers and can use the "2D Nav Goal" tool to set and send target points. Would you like to ask if the output of the trained model is only linear and angular velocity? If so, would it be a real world test if I could take the model trained on the Turtlebot 2 and put it to the test on the Ackerman chassis? The following is the problem I encountered when testing, and the error message in red is what I mentioned above. Thank you for your guidance and help.
[image]
[image]
[image]
[image]

zzuxzt, 2024-10-25:
The output commands are the linear velocity and angular velocity (Vx, Wz). It should also be deployed on your Ackerman robot as it has been tested in Turtlebot 2 and Jackal robots. 

The problem may be caused by an inappropriate goal point. You can replace the 2nd goal point or remove it.

GGHaozhu, 2024-11-05:
好的，谢谢您

GGHaozhu, 2024-11-05:
Ok, thank you for your reply.

closed 2024-11-06 by zzuxzt