TempleRAIL/drl_vo_nav #31: When running the DRL-VO inference code, the robot keeps swaying left and right along the global path.

yimengweima0, 2024-10-31:
Compared to your demo, I only removed AMCL and used the odometry ground truth instead.  The ouytput is as follows:
started roslaunch server http://localhost:46831/

SUMMARY
========

PARAMETERS
 * /drl_vo_cmd/model_file: /home/lwh/G314/co...
 * /move_base/DWAPlannerROS/acc_lim_theta: 2.0
 * /move_base/DWAPlannerROS/acc_lim_x: 1.0
 * /move_base/DWAPlannerROS/acc_lim_y: 1.0
 * /move_base/DWAPlannerROS/forward_point_distance: 0.5
 * /move_base/DWAPlannerROS/global_frame_id: odom
 * /move_base/DWAPlannerROS/goal_distance_bias: 24.0
 * /move_base/DWAPlannerROS/max_scaling_factor: 0.2
 * /move_base/DWAPlannerROS/max_vel_rot: 3.0
 * /move_base/DWAPlannerROS/max_vel_trans: 0.5
 * /move_base/DWAPlannerROS/max_vel_x: 0.5
 * /move_base/DWAPlannerROS/max_vel_y: 0.0
 * /move_base/DWAPlannerROS/min_vel_rot: 0.4
 * /move_base/DWAPlannerROS/min_vel_trans: 0.1
 * /move_base/DWAPlannerROS/min_vel_x: 0.0
 * /move_base/DWAPlannerROS/min_vel_y: 0.0
 * /move_base/DWAPlannerROS/occdist_scale: 0.5
 * /move_base/DWAPlannerROS/oscillation_reset_dist: 0.05
 * /move_base/DWAPlannerROS/path_distance_bias: 64.0
 * /move_base/DWAPlannerROS/publish_cost_grid_pc: True
 * /move_base/DWAPlannerROS/publish_traj_pc: True
 * /move_base/DWAPlannerROS/scaling_speed: 0.25
 * /move_base/DWAPlannerROS/sim_time: 1.0
 * /move_base/DWAPlannerROS/stop_time_buffer: 0.2
 * /move_base/DWAPlannerROS/theta_stopped_vel: 0.4
 * /move_base/DWAPlannerROS/trans_stopped_vel: 0.1
 * /move_base/DWAPlannerROS/vtheta_samples: 20
 * /move_base/DWAPlannerROS/vx_samples: 6
 * /move_base/DWAPlannerROS/vy_samples: 1
 * /move_base/DWAPlannerROS/xy_goal_tolerance: 0.9
 * /move_base/DWAPlannerROS/yaw_goal_tolerance: 6.28
 * /move_base/GlobalPlanner/allow_unknown: True
 * /move_base/GlobalPlanner/cost_factor: 3.0
 * /move_base/GlobalPlanner/default_tolerance: 0.0
 * /move_base/GlobalPlanner/lethal_cost: 253
 * /move_base/GlobalPlanner/neutral_cost: 50
 * /move_base/GlobalPlanner/old_navfn_behavior: False
 * /move_base/GlobalPlanner/planner_costmap_publish_frequency: 0.0
 * /move_base/GlobalPlanner/planner_window_x: 0.0
 * /move_base/GlobalPlanner/planner_window_y: 0.0
 * /move_base/GlobalPlanner/publish_potential: True
 * /move_base/GlobalPlanner/publish_scale: 100
 * /move_base/GlobalPlanner/use_dijkstra: True
 * /move_base/GlobalPlanner/use_grid_path: False
 * /move_base/GlobalPlanner/use_quadratic: True
 * /move_base/NavfnROS/allow_unknown: False
 * /move_base/NavfnROS/default_tolerance: 0.3
 * /move_base/NavfnROS/planner_window_x: 0.0
 * /move_base/NavfnROS/planner_window_y: 0.0
 * /move_base/NavfnROS/visualize_potential: False
 * /move_base/base_global_planner: navfn/NavfnROS
 * /move_base/base_local_planner: dwa_local_planner...
 * /move_base/controller_frequency: 5.0
 * /move_base/controller_patience: 3.0
 * /move_base/global_costmap/global_frame: map
 * /move_base/global_costmap/inflation_layer/cost_scaling_factor: 5.0
 * /move_base/global_costmap/inflation_layer/enabled: True
 * /move_base/global_costmap/inflation_layer/inflation_radius: 0.5
 * /move_base/global_costmap/map_type: voxel
 * /move_base/global_costmap/max_obstacle_height: 0.6
 * /move_base/global_costmap/obstacle_layer/bump/clearing: False
 * /move_base/global_costmap/obstacle_layer/bump/data_type: PointCloud2
 * /move_base/global_costmap/obstacle_layer/bump/marking: True
[... 125 lines cut ...]
process[mix_cmd_vel-12]: started with pid [56066]
process[rviz-13]: started with pid [56074]
[ WARN] [1730386880.361910773, 1288.350000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: Lookup would require extrapolation at time 1288.350000000, but only time 1288.450000000 is in the buffer, when looking up transform from frame [base_footprint] to frame [map]. canTransform returned after 1288.35 timeout was 0.1.
[ WARN] [1730386880.473006125, 1288.460000000]: global_costmap: Pre-Hydro parameter "static_map" unused since "plugins" is provided
[ WARN] [1730386880.473401133, 1288.460000000]: global_costmap: Pre-Hydro parameter "map_type" unused since "plugins" is provided
[ INFO] [1730386880.474248454, 1288.460000000]: global_costmap: Using plugin "static_layer"
[ INFO] [1730386880.483553402, 1288.470000000]: Requesting the map...
[ INFO] [1730386880.716775047, 1288.670000000]: Resizing costmap to 1376 X 1376 at 0.050000 m/pix
[ INFO] [1730386880.835991846, 1288.770000000]: Received a 1376 X 1376 map at 0.050000 m/pix
[ INFO] [1730386880.844868796, 1288.770000000]: global_costmap: Using plugin "obstacle_layer"
[ INFO] [1730386880.859640216, 1288.780000000]:     Subscribed to Topics: scan bump
[ INFO] [1730386880.952342692, 1288.870000000]: global_costmap: Using plugin "inflation_layer"
[ WARN] [1730386881.029227573, 1288.930000000]: local_costmap: Pre-Hydro parameter "static_map" unused since "plugins" is provided
[ WARN] [1730386881.029648502, 1288.930000000]: local_costmap: Pre-Hydro parameter "map_type" unused since "plugins" is provided
[ INFO] [1730386881.030432415, 1288.930000000]: local_costmap: Using plugin "obstacle_layer"
[ INFO] [1730386881.039993145, 1288.940000000]:     Subscribed to Topics: scan bump
[ INFO] [1730386881.092185828, 1288.990000000]: local_costmap: Using plugin "inflation_layer"
[ INFO] [1730386881.138991123, 1289.040000000]: Created local_planner dwa_local_planner/DWAPlannerROS
[ INFO] [1730386881.147251580, 1289.040000000]: Sim period is set to 0.20
[ INFO] [1730386881.340185560, 1289.220000000]: Recovery behavior will clear layer 'obstacles'
[WARN] [1730386881.354703, 1289.240000]: Could not get robot pose
[ INFO] [1730386881.356986664, 1289.240000000]: Recovery behavior will clear layer 'obstacles'
[WARN] [1730386881.366097, 1289.250000]: Could not get robot pose
[WARN] [1730386881.405211, 1289.290000]: Could not get robot pose
[ INFO] [1730386881.439097502, 1289.310000000]: odom received!

yimengweima0, 2024-10-31:
just like this 
https://github.com/user-attachments/assets/13234c0f-9ec3-4c40-a9b0-f74db425e92a

yimengweima0, 2024-11-01:
Could you please assist me in resolving them? I would really appreciate your guidance. Thank you!" @zzuxzt

zzuxzt, 2024-11-01:
Modify lines 133-134 of [drl_vo_inference.py](https://github.com/TempleRAIL/drl_vo_nav/blob/baa970c30cabe94a8b0eb81bb0ad0b97261f2458/drl_vo/src/drl_vo_inference.py#L133C1-L134C29) as follows::
`
vz_min = -0.7
`
`
vz_max = 0.7
`

yimengweima0, 2024-11-01:
ok, thank you

closed 2024-11-06 by zzuxzt