TempleRAIL/drl_vo_nav #33: Sensor topic /cnn_data is not available，Sensor topic /map is not available Couldn't find the left cliff sensor in the model!

panzhangchao, 2024-11-30:
[image]
[image]

When I run "sh run_drl_vo_policy_training_server.sh ~/drl_vo_runs", I encountered the error shown in the above image, but when I ran "rostopin info/cnn.data", the following image appeared. May I ask if this is related to the network.
[image]

Thanks very much

closed 2024-12-01 by panzhangchao

DavidYan628, 2025-01-21:
Hello, I encountered the same problem when I ran "sh run_drl_vo_policy_training_server.sh ~/drl_vo_runs". How did you solve it?

HS678, 2025-03-11:
> Hello, I encountered the same problem when I ran "sh run_drl_vo_policy_training_server.sh ~/drl_vo_runs". How did you solve it?

I also encountered the problem.Do you solve it?And how to solve ?

shaashou, 2025-08-05:
I also encountered the problem.Do you solve it?And how to solve ?