TempleRAIL/drl_vo_nav #38: Problem on container version installation

Weizhi-T, 2025-01-21:
Dear Zhanteng,

Thank you so much for your great work on the DRL-VO. I am using your container version.
By following your instruction of installation, I am trying to run your training and inference     .sh  file.

But the termination reports the issue that: ModuleNotFoundError: No module named 'zipp'. The proccess died. 
And: futher reports: Could not get robot pose
And: (gzclient:162691): dconf-CRITICAL **: 20:48:15.324: unable to create directory '/run/user/1000/dconf': Read-only file system.  dconf will not work properly.
The detail is showed below:

[image]

[image]

zzuxzt, 2025-01-23:
You can try to install the 'zipp' in the container:
`pip install zipp`

Weizhi-T, 2025-01-23:
I have already tried. It reported: AttributeError: module 'lib' has no attribute 'X509_V_FLAG_NOTIFY_POLICY'

[image]

I thought it might be the problem of version of the OpenSSL. I tried to update: sudo apt install -y openssl

[image]

still not working. Is there any other ways I could try or something I missed. Thank you!

zzuxzt, 2025-01-23:
What is your system version? It should be lower than Ubuntu 20.04. I know higher versions have some issues with the current container.

Weizhi-T, 2025-01-25:
It is my system information:
         Icon name: computer-desktop
           Chassis: desktop
  Operating System: Ubuntu 20.04.6 LTS
            Kernel: Linux 5.15.5-051505-generic
      Architecture: x86-64

zzuxzt, 2025-03-05:
You can try installing the zipp and openssl packages in your Ubuntu system, reboot, restart the container, and then restart the compilation again. Or try it in a new computer?

hufenger712, 2025-04-28:
Have you solved this problem? I encountered the same issue. How did you resolve it?