TempleRAIL/drl_vo_nav #45: Problem with training

BRoln7, 2025-04-23:
I would like to know if this policy is trained using PPO after being randomly initialized in Gazebo? If so, how many steps does it roughly take to converge, and how long does the training time take?

zzuxzt, 2025-04-29:
Yeah, in my case, it took at least 1M steps and about a week to ten days to train it on the server. More training. Please refer to https://github.com/TempleRAIL/drl_vo_nav/issues/39#issuecomment-2609386810 and https://github.com/TempleRAIL/drl_vo_nav/issues/25#issuecomment-2087934233 for more training experience.

BRoln7, 2025-04-29:
Thank you for your reply! It is very helpful and provides valuable reference for my upcoming DRL training.