TempleRAIL/drl_vo_nav #47: What is the sub-goal point?

HS678, 2025-05-20:
Hello, first of all, thank you very much for your DRL-VO research, I am now a little confused about the sub-goal point in the input, is this sub-goal point different from the target point clicked in the Rviz map, is it the process of the robot reaching the final goal point will set a series of sub-goal points by itself?

zzuxzt, 2025-05-23:
Thank you for your interest in our work. Note that the target point clicked in the Rviz is the final goal point, while the goal input of our DRL-VO is the sub-goal point generated by the pure-pursuit algorithm. More details on generating sub-goal points can be found in our [paper](https://doi.org/10.1109/TRO.2023.3257549).

HS678, 2025-05-23:
OK，thank you vary much.

closed 2025-08-08 by zzuxzt